borti4938/SNES_MultiRegion_with_DeJitter_QID #2: DFO programming via Arduino verify fails.

ArcadeTV, 2020-11-30:
I get this error on both UNO and NANO boards, testing with 2 different PCBs.
Of course this could be a user issue.
Value read at reg 0x1E: 0x2 (0x72 expected) looks suspicious.

This is my first SNES_MultiRegion_with_DeJitter_QID install and I wanted to say thanks and give a big hug!

```
Wait for CDCE913: ... Found! (CDCE913 I2C Address: 0x65)
Write registers: ................................ done!
Verify registers: ... 
 - Number of bytes returned: 32
 - Value read at reg 0x0: 0x81 (not evaluated as reg is read only)
 - Value read at reg 0x1: 0x1 (0x1 expected)
 - Value read at reg 0x2: 0xBC (0xBC expected)
 - Value read at reg 0x3: 0x6 (0x6 expected)
 - Value read at reg 0x4: 0x1 (0x1 expected)
 - Value read at reg 0x5: 0x90 (0x90 expected)
 - Value read at reg 0x6: 0x40 (0x40 expected)
 - Value read at reg 0x7: 0x40 (not evaluated as reg is read only)
 - Value read at reg 0x8: 0x40 (not evaluated as reg is read only)
 - Value read at reg 0x9: 0xCF (not evaluated as reg is read only)
 - Value read at reg 0xA: 0xFF (not evaluated as reg is read only)
 - Value read at reg 0xB: 0xFF (not evaluated as reg is read only)
 - Value read at reg 0xC: 0xFF (not evaluated as reg is read only)
 - Value read at reg 0xD: 0xFF (not evaluated as reg is read only)
 - Value read at reg 0xE: 0xFF (not evaluated as reg is read only)
 - Value read at reg 0xF: 0xFF (not evaluated as reg is read only)
 - Value read at reg 0x10: 0x0 (0x0 expected)
 - Value read at reg 0x11: 0x0 (0x0 expected)
 - Value read at reg 0x12: 0x0 (0x0 expected)
 - Value read at reg 0x13: 0x1 (0x1 expected)
 - Value read at reg 0x14: 0x6B (0x6B expected)
 - Value read at reg 0x15: 0x1 (0x1 expected)
 - Value read at reg 0x16: 0x18 (0x18 expected)
 - Value read at reg 0x17: 0x5 (0x5 expected)
 - Value read at reg 0x18: 0x4F (0x4F expected)
 - Value read at reg 0x19: 0x55 (0x55 expected)
 - Value read at reg 0x1A: 0x53 (0x53 expected)
 - Value read at reg 0x1B: 0xEC (0xEC expected)
 - Value read at reg 0x1C: 0x96 (0x96 expected)
 - Value read at reg 0x1D: 0xF1 (0xF1 expected)
 - Value read at reg 0x1E: 0x2 (0x72 expected)
... VERIFICATION ERROR!
Number of Bytes not as expected!
First register error at 0x1E!
```

ArcadeTV, 2020-12-01:
[image]

Getting the same failure with any combination of arduinos and dfo's.

borti4938, 2020-12-01:
I'm not very experienced with Arduino either. So I'm not quite sure what's exactly going on. Just a broad guess, which might be completely wrong: It seems like that the total number of values read and write with the Wire library is limited. (which was not the case when I tested it..???).
Could you please try the following implementation where I cut both writes and reads in half pieces.

[Test code for Alex](https://1drv.ms/u/s!AosQhmNoM7qSjVq8mv0kHfOZXwAa?e=feOmjE)

Not sure if it runs correctly as I cannot test it at the moment.

Good luck!

ArcadeTV, 2020-12-01:
That did the trick :-)
It's soo nice to have this alternative for programming!
Thanks my dear!

```
Wait for CDCE913: ... Found! (CDCE913 I2C Address: 0x65)
Write registers: ................................ done!
Verify registers: ... 
 - Value read at reg 0x0: 0x81 (not evaluated as reg is read only)
 - Value read at reg 0x1: 0x1 (0x1 expected)
 - Value read at reg 0x2: 0xBC (0xBC expected)
 - Value read at reg 0x3: 0x6 (0x6 expected)
 - Value read at reg 0x4: 0x1 (0x1 expected)
 - Value read at reg 0x5: 0x90 (0x90 expected)
 - Value read at reg 0x6: 0x40 (0x40 expected)
 - Value read at reg 0x7: 0x40 (not evaluated as reg is read only)
 - Value read at reg 0x8: 0x40 (not evaluated as reg is read only)
 - Value read at reg 0x9: 0x40 (not evaluated as reg is read only)
 - Value read at reg 0xA: 0x0 (not evaluated as reg is read only)
 - Value read at reg 0xB: 0x0 (not evaluated as reg is read only)
 - Value read at reg 0xC: 0x0 (not evaluated as reg is read only)
 - Value read at reg 0xD: 0x0 (not evaluated as reg is read only)
 - Value read at reg 0xE: 0x0 (not evaluated as reg is read only)
 - Value read at reg 0xF: 0x0 (not evaluated as reg is read only)
 - Value read at reg 0x10: 0x0 (0x0 expected)
 - Value read at reg 0x11: 0x0 (0x0 expected)
 - Value read at reg 0x12: 0x0 (0x0 expected)
 - Value read at reg 0x13: 0x1 (0x1 expected)
 - Value read at reg 0x14: 0x6B (0x6B expected)
 - Value read at reg 0x15: 0x1 (0x1 expected)
 - Value read at reg 0x16: 0x18 (0x18 expected)
 - Value read at reg 0x17: 0x5 (0x5 expected)
 - Value read at reg 0x18: 0x4F (0x4F expected)
 - Value read at reg 0x19: 0x55 (0x55 expected)
 - Value read at reg 0x1A: 0x53 (0x53 expected)
 - Value read at reg 0x1B: 0xEC (0xEC expected)
 - Value read at reg 0x1C: 0x96 (0x96 expected)
 - Value read at reg 0x1D: 0xF1 (0xF1 expected)
 - Value read at reg 0x1E: 0x72 (0x72 expected)
 - Value read at reg 0x1F: 0x69 (0x69 expected)
... done!
Store registers in EEPROM: ... requested ... done!
SUCCESS!
```

borti4938, 2020-12-01:
thank you for your feedback 👍 
fix is in commit 0936e4b19fc30c70eed8a8d232697d76a9ff9820

closed 2020-12-01 by borti4938