borti4938/SNES_MultiRegion_with_DeJitter_QID #3: Flashing the CPLD with 3.3V

ArcadeTV, 2020-12-01:
This is more of a question than an issue...
I have an Altera USB-Blaster that I kept for flashing updated firmwares to the nesrgb. On my DIY-adapter I connect a 3.3V power supply.
Since you mentioned "please use the 5V power supply rail to not harm the DC converters" in the readme I'm unsure if it's safe to use the 3.3V instead of 5V while the pcb is not installed into the console yet.

ArcadeTV, 2020-12-01:
[image]

The flashing was reported successful, while powering with 3.3V and pcb not installed yet.

borti4938, 2020-12-01:
Personally I power the board over the I2C header of the CDCE913. It is 5V 'global' board power there.
By doing so, you will have the reference voltage of the CPLD set and can use the USB Blaster.

Powering the board just over the JTAG header might dagerous as you are 'behind' the 3.3V LDO. I cannot recommend that.

closed 2020-12-01 by borti4938

ArcadeTV, 2020-12-01:
I will do so, thanks for clearing that up!