borti4938/SNES_MultiRegion_with_DeJitter_QID #4: Documentation help offer

ArcadeTV, 2020-12-04:
I had a hard time figuring out all possible combinations of parts and connections when using the SMR20200325 and SNES_RGBAmp_wCSYNC, so I made this flowChart:

https://www.dropbox.com/s/7jbngqexuoay9ys/FlowChart_-_SMR20200325_and_SNES_RGBAmp_wCSYNC.pdf?dl=1

If you find the time to review it and leave some feedback for improvements, I'd like to offer to make a chart for all SNES models, so you can add them to the repository for others that may have trouble gathering all the info on both projects.

ArcadeTV, 2020-12-04:
I just updated the PDF to fix some typos and fixed some alignment.

ArcadeTV, 2020-12-06:
Also I found that google forms is a great tool for conditional steps of an install guide.

https://forms.gle/T5UTCCGdLv6wvKRK8 (mockup, incomplete)

duncan1111, 2021-05-25:
on your documentation, when you talk about the RGB bypass. You say:
NTSC > Populate R13,R23,R33 with 39Ω
PAL > Populate R13,R23,R33 with 75Ω

Should this not be the other way around, in the BOM its 75Ω for NTSC and 39Ω for PAL?

borti4938, 2024-04-24:
Thank you my lovely Alex :)

closed 2024-04-24 by borti4938