borti4938/SNES_MultiRegion_with_DeJitter_QID #5: DFO programmer 5V?

whf789, 2021-06-01:
When programming the DFO, the readme says "Set V-target jumper on programmer to 5V"
(here: https://github.com/borti4938/SNES_MultiRegion_with_DeJitter_QID/blob/master/fw/dfo/SOURCE.md)

I am using this DFO programmer https://oshpark.com/shared_projects/SeJLCY52 which is recommended, but it only outputs 3.3V. How did you modify the DFO programmer output 5V for this process?

Thanks for this project! 👍

borti4938, 2021-06-02:
Hey :)

If I were you I would simply test with 3.3V first. If you are lucky it works. At least you cannot break anything.
If it does not work, you can get 5V from the 100nF ceramic capacitor right next to the FT232 chip (lower pad). Make sure that the slider is turned off and then connect 5V from the cap with Vtg (LED should turn on although slider is off).

Best Regards and good luck :)

whf789, 2021-06-02:
Hi Borti,

I have tried with 3.3V and it said "CDCE913 successfully programmed" in the CDCEXX Programmer, so I assume it worked :)

Thanks for your reply & help.

closed 2021-06-02 by whf789

borti4938, 2021-06-02:
Great 👍 Good luck with installation and have fun with the mod 😃