borti4938/SNES_MultiRegion_with_DeJitter_QID #6: Feature Request: Enable/Disable dejitter with ulGR

FixingGlobe, 2021-09-01:
Adding a sliding switch is not always an elegant solution to enable/disable dejitter via SJ31.
I think it should be possible to add a transistor to enable/disable dejitter. SJ31 should still remain because of backwards compatibility. Of course it will need a circuit to prevent SJ31 and the transistor to both be enabled and disabled at the same time. In any case if SJ31 is closed, this should count as master. This will require a new pcb.
For the button combination I would suggest the retrobit combination and switch Start with Select.
This is mostly for speedrunners and I even try to come up with a circuit and software by myself.

borti4938, 2022-04-15:
This needs redesign as the PIC does not have a pin free for that. Sorry - no chance at the moment. But it is on my todo since I was designing this board. But my personal interest was not high enough ;)

borti4938, 2024-04-24:
May never happen... Closed for the moment

closed 2024-04-24 by borti4938