borti4938/SNES_MultiRegion_with_DeJitter_QID #7: CPLD (U3) replacement?

whf789, 2022-03-02:
Hello,

Is there any suitable replacement for U3 = EPM7032AETC44-10N ?

The chip is now obsolete.

Thanks!

borti4938, 2022-03-29:
As far as I see it, the ATF15xx series by microchip should be a drop in replacement.
There are 3.3V Vcc varaints (ATF1502ASV), which can be used directly, and 5V variants (ATF1502AS), which need a slight PCB modification.
I bought some ATF1504ASV (ATF1502ASV was not available) and see how it goes. Afterwards I can update instructions.

duncan1111, 2022-03-30:
https://www.digikey.co.uk/en/products/detail/microchip-technology/ATF1504ASVL-20AU44/1008357

this one?

borti4938, 2022-03-31:
> https://www.digikey.co.uk/en/products/detail/microchip-technology/ATF1504ASVL-20AU44/1008357
> 
> this one?

exactly. They arrived today. I will let you know once I found the time.

> Now to find some CDCE913... They're out of stock everywhere and there seems to be no suitable replacement without changing the package.

Actually, I don't know if I have some spares at home. But I defenitely have some spare CPLDs (ten or so) if this helps.

duncan1111, 2022-03-31:
Is the "marqs85" dejitter code the same as your one?

I ask as i want to know if i can take the ATF1502AS off a marqs85 snes dejitter and install on the AIO cic board?

I also have some EPM7032STC44-10N. are these suitable also?

borti4938, 2022-04-03:
> I ask as i want to know if i can take the ATF1502AS off a marqs85 snes dejitter and install on the AIO cic board?

Unfortunately, marqs85 uses the 5V variant, I use the 3.3V variant as I initially went for the Altera CPLDs, which were only available in 3.3V. Using the 5V variants would need to modify the PCB a bit.

> EPM7032STC44

Same here, also a 5V chip.

> exactly. They arrived today. I will let you know once I found the time.

Good news: I successfully tested one :) Just have to write instructions - maybe this evening.

borti4938, 2022-04-03:
fw builds and instructions are ready

closed 2022-04-03 by borti4938