borti4938/SNES_MultiRegion_with_DeJitter_QID #8: SHVC-CPU-01 installation

RamsesUTL, 2022-04-14:
Hi ... I bought the mod from VGP and did not check upfront which board version I have (stupid I know).
Turns out I have a first generation US SNES (with the discrete audio card) and the SHVC-CPU-01 board and there is no installation guide listed here for that :-(
Because it was my first own console back in the days I'm emotionally attached to that US device and would really love to use it with the mod, but I'm not sure if I'm good enough to just follow the general installation instructions and figure everything out by myself.

Has anyone experience with the installation in that SNES model and can provide instructions or pictures?
Or is there maybe someone in Germany that could offer installation service?
Thanks!

borti4938, 2022-04-14:
OK, good news I guess. I also bought the mod from VGP and successfully modded today a SHVC-CPU-01 board today. I think I will have the description ready in a few days. So you should be good to go then such that you can try it on your own. :) Otherwise I can also offer installation service. Just contact me if you like to take that offer.
Best regards :)

RamsesUTL, 2022-04-14:
You are a lifesaver. Thanks for the offer. I'll take a look at your instructions and if I'm not starting to sweat right away I'll give it a try. Otherwise you'll get a cry for help mail from me ;-)

borti4938, 2022-04-15:
Here you are :) [Installation for SHVC-CPU-01](https://github.com/borti4938/SNES_MultiRegion_with_DeJitter_QID/tree/master/installation/SHVC-CPU-01)

closed 2022-04-15 by borti4938