borti4938/SNES_MultiRegion_with_DeJitter_QID #12: New Atmel CPLD Issue

whf789, 2022-06-25:
Hi,
I am trying to program the new ATMEL CPLD part no. ATF1504ASV(L) as per the instructions on the GitHub page but am having a bit of trouble. 

When initializing the jtag connection, I get the error: 

`C:\Users\X>openocd -f openocd-ft2232h.conf
Open On-Chip Debugger 0.11.0
Licensed under GNU GPL v2
For bug reports, read
        http://openocd.org/doc/doxygen/bugs.html
C:/msys64/mingw64/bin/../share/openocd/scripts/openocd-ft2232h.conf:3: Error: invalid command name "ftdi"
in procedure 'script'
at file "embedded:startup.tcl", line 26
at file "C:/msys64/mingw64/bin/../share/openocd/scripts/openocd-ft2232h.conf", line 3`

I was told elsewhere to "Not to use some out-of-tree way to init JTAG adapter and instead source ftdi/minimodule.cfg."

Moreover, I was also told "This odd script isn't suitable for this version of OpenOCD, use scripts that come with the binary you got there."

Any help would be appreciated, I must say it was much more straightforward when I programmed two boards in the past using the EPM7032.

borti4938, 2022-06-25:
Can you use a build later than July 29th 2021? At least this version works for me. Afaik some commands were changed.

This is my first project where I worked with opencd. I didn't want to write "please go and buy an expensive Atmel programmer" in the documentation.

Side note:

> I was told elsewhere to "Not to use some out-of-tree way to init JTAG adapter and instead source ftdi/minimodule.cfg."
>
> Moreover, I was also told "This odd script isn't suitable for this version of OpenOCD, use scripts that come with the binary you got there."

I know that I'm not an expert, but at least I think that I'm somehow not a noob. I cannot understand what the two trolls want you to do. The first one is somehow ok, but he should have mentioned where to source and what to do then (use out of box, some adaptations needed, what ever)

The second is just a troll. Sounds to me: "The script is odd or just crap, but I don't say why..."

whf789, 2022-06-26:
Borti, thanks for your reply.

Firstly please allow me to apologize. I simply copied and pasted their reply as I thought it may somewhat be helpful incase there was a problem with the .conf file, but it seems that is not the problem at all.

Back to the issue on hand... I followed your advice as I was previously using the "latest"/stable release 07 Mar 2021 v0.11.0. After updating to the pre-release from two days ago, it seems to have now detected successfully!

However, when following your instructions to open another terminal to interact with openocd:

`telnet localhost 4444
> svf <full_path_to_svf_file>`

I cannot get it to work. Here is what it says on my command prompt:

Command prompt to initialize JTAG: 

`C:\Users\X>openocd -f openocd-ft2232h.conf
Open On-Chip Debugger 0.11.0+dev-00748-ga70bab9e5 (2022-06-24-21:57)
Licensed under GNU GPL v2
For bug reports, read
        http://openocd.org/doc/doxygen/bugs.html
adapter speed: 1000 kHz

Info : Listening on port 6666 for tcl connections
Info : Listening on port 4444 for telnet connections
Info : clock speed 1000 kHz
Warn : There are no enabled taps.  AUTO PROBING MIGHT NOT WORK!!
Info : JTAG tap: auto0.tap tap/device found: 0x0151403f (mfg: 0x01f (Atmel), part: 0x1514, ver: 0x0)
Warn : AUTO auto0.tap - use "jtag newtap auto0 tap -irlen 3 -expected-id 0x0151403f"
Warn : gdb services need one or more targets defined
Info : accepting 'telnet' connection on tcp/4444
svf [-tap device.tap] <file> [quiet] [nil] [progress] [ignore_error]
xsvf (tapname|'plain') filename ['virt2'] ['quiet']`

telnet localhost 4444 command prompt:

`C:\Users\Will>telnet localhost 4444
Trying ::1...
Connection failed: Connection refused
Trying 127.0.0.1...
Connected to localhost.
Escape character is '^]'.
Open On-Chip Debugger
> svf <C:\Users\X\Downloads\SNES_MultiRegion_with_DeJitter_QID-master (3)\SNES_MultiRegion_with_DeJitter_QID-master\fw\Logic\output_files\multi_func_atmel\multi_function_64.svf>
svf [-tap device.tap] <file> [quiet] [nil] [progress] [ignore_error]
xsvf (tapname|'plain') filename ['virt2'] ['quiet']`

Any ideas what to try next? I cannot get it to initiate the programming.

whf789, 2022-07-17:
Worked this out btw.
If anyone using windows has the same issue, before openocd, cd to the path then use svf <filename> directly and it will work.

closed 2022-07-17 by whf789

borti4938, 2022-07-18:
Sorry that I missed that. Glad that you found a solution for you. Otherwise you have to use the slash " /" in the path name instead of backslash "\" when connected to the OpenOCD although you are running it on Windows.