borti4938/SNES_MultiRegion_with_DeJitter_QID #13: BOM Substitutions

meisenbarth, 2022-07-19:
Is the CDCEL913 a valid substitute for CDCE913? The main differences appears to be Vddout max. 

I'm also assuming the PIC16F18324 will be a valid sub for the older PIC16F684.

borti4938, 2022-07-21:
You cannot use the CDCEL913 as VDDOUT is connected to 3.3V.

For the 16F18324: you can go ahead and port the code. The hex file for the 16F684 will probably not work.

Razer41, 2022-08-02:
A follow up question on BOM substitutions if that's ok... Is there any drop in replacements for the following?
TSH122ICT
9C-27.000MAAJ-T

borti4938, 2022-08-07:
For thd TSH122ICT you can use the OPA360AIDCKR. For the 9C-27.000MAAJ-T you may use the HC49USM-FF3F18-27.000MHZ or HC49USM-FF5F18-27.000MHZ.

I should update the BOM, I know

meisenbarth, 2022-09-17:
For the TSH122ICT can you confirm the part orientation on the newest board (pin1)? I tried to view the brd file and was unable to get it to load.

borti4938, 2022-09-17:
Pin 1 is bottom left

~~~~
        +-------+
R92 <-  | 6 5 4 |  -> to C93
SJ93 <- | 1 2 3 |  -> to C94
        +-------+
~~~~

Razer41, 2022-09-21:
There is no other substitutions for ATF1502ASV or ATF1504ASV(L)? Both sadly out of stock until next years.
Only package remaining for this parts is PLCC variant which has different pin pitch so cannot be used

borti4938, 2022-09-25:
It is possbily suitable to replace the CPLD with a XCR3064XL. But it is also difficult to source one, here.

Scenz, 2023-01-27:
@borti4938 do you think a EPM3064A could also work as they are good available and also 5V tolerant

borti4938, 2023-01-28:
Might be possible, but you have to change the PCB layout first