borti4938/SNES_MultiRegion_with_DeJitter_QID #17: Controller sniff suddenly stopped working

kirkegaard, 2024-03-26:
Hi, Ive had my board installed for a couple of weeks now and its been working great. But today while playing the key combinations stopped working. It wont let me region switch, or reset the console. The auto region switch still works and as i understand it, the uIGR is the one that controls that AND the input sniff? Is it possible to disable the input sniff accidentally maybe? Or do you have any other clues for what i need to check? :)

borti4938, 2024-04-01:
https://github.com/borti4938/SNES_MultiRegion_with_DeJitter_QID/blob/master/fw/uIGR/Manuals/uIGR_Manual.pdf

Page no. 3 - lock types

kirkegaard, 2024-04-01:
Oh my god! That was it! Thank you thank you thank you!

Ive been poking around for at least one week now and couldnt figure out what i did wrong. I cant believe i missed the lock types! The board has been working great on my SNSP-CPU-02 SNES! Thank you for all your work on this board

closed 2024-04-01 by kirkegaard