borti4938/SNES_MultiRegion_with_DeJitter_QID #18: SuperCIC + FXPAK PRO pairing mode not working

DerElefant69, 2024-04-20:
Hey, first of all thanks for this awesome work!

I have a PAL SNES with a SNSP-CPU-02 mainboard. Yesterday I put in the modchip according to the documentation on this site and everything seems to be okay. My PAL cartridges work fine and the modchip is detecting the Game as PAL 50Hz in Auto Mode (unfortunately I don't have NTSC cartridges to test with). I can also switch the modes manually with the reset button and with the Controller Inputs that come with the uIGR, so everything ok.

Now with my FXPAK PRO on the latest Firmware there is a strange behavior (didnt test other firmware). I have done all settings as described to turn on the pairing mode for the SuperCIC and the FXPAK PRO. Now when I turn on the console, I get 60Hz, but when I start a PAL rom it changes to 50Hz for some miliseconds and then jumps back to 60Hz. I can't switch the mode with the reset button anymore, only with Controller Inputs over uIGR, but this should normally not be necessary in pairing mode? Do you have any idea why this happens? I didn't find anything on the net and any help would be appreciated. 

- I tested several PAL roms
- I turned off auto region patch in the FXPAK PRO
- Region patch from SuperCIC is active
- Turned on the SuperCIC mode for uIGR as described.

I checked the values of the frequency over a Retrotink to be sure. 

Thanks in advance and greetings from Germany 🙂

PS: I bought the modchip from Retroupgrades.co.uk

borti4938, 2024-04-24:
Hi,

it is normal that switching modes via reset button does not work anymore as soon as the SuperCIC is in pairmode.
The other issue: I assume that you set the menu settings to auto. It sounds like that lock and key run out of sync somehow. But I'm not really sure about that. The only thing which you can check, is if the SuperCIC on the mod board is properly soldered (incl. clock line and pins on the cart slot pins 56, 55, 25 and 24). Also you may give the edge connector of the cart slor a clean...

Maybe it is also something completely different. One question: does the region patching works on PAL games? Please try e.g. MMX2 or MMX3 on PAL (playing through first level - if the game resets the region patching assumes a NTSC ROM being loaded...

Best regards (also from Germany)

DerElefant69, 2024-04-26:
Hi,

yes, the setting in the SuperCIC menu is set to auto.

I tried several things now, but I'm not getting any closer to solve the problem. Here are the things I tried:

- cleaned the cartridge connector and the cartridge pins with IPA
- resoldered every connection from the mod board to the cartridge pins
- measured continuity from the mod board cartridge connection to the outer pins of the cartridge connector -> OK
- measured continuity from mod board SuperCIC pin 14 to Original CIC pad 9 and cartridge slot pin 5 + 36-> OK
- measured continuity from mod board SuperCIC pin 13 to Original CIC pad 8 -> OK
- measured continuity from mod board SuperCIC pin 11 to Original CIC pad 10-> OK
- measured continuity from mod board SuperCIC pin 10 to Original CIC pad 1 and cartridge slot pin 55 -> OK
- measured continuity from mod board SuperCIC pin 9 to Original CIC pad 2 and cartridge slot pin 24 -> OK
- measured continuity from mod board SuperCIC pin 8 to Original CIC pad 11 and cartridge slot pin 25 -> OK
- measured continuity from mod board SuperCIC pin 2 to Original CIC pad 7 -> OK 
- measured continuity from mod board SuperCIC pin 2 to cartridge slot pin 56 -> NOT OK(?) (because jumper SJ11 is open. I soldered the Clk.CIC to pin 6 from S-CLK, as described in the installation instructions) 
- measured continuity from mod board SuperCIC pin 1 to Original CIC pad 18 and cartridge slot pin 27 + 58 -> OK
- tried to close jumper SJ11 without soldering Clk.CIC to pin 6 of S-CLK in the first place, but that didn't work either.

I also tried MMX2 NTSC ROM on FXPAK pro (with deactivated region patch in settings) with fixed PAL mode:

- mod board region patch activated: ROM is playing as expected in 50Hz, without warning that the game pak is not for PAL region and I also get further than the first level.

- mod board region patch deactivated: ROM starts with warning, that game pak is not for PAL region, and after the first level the game gets reset.

So I guess that the region patch is working fine.

And now, after I spent some hours, I run out of ideas :) Maybe you have another idea why this happens. If not, I will live with it as it is :)

jamieshaw, 2024-05-14:
I too have a similar issue with an SNSP-CPU-02 board, however with a SD2SNES Rev. X cart (I have an official FXPAK PRO on the way).

It seems that the pair mode is somewhat of [a known issue](https://github.com/mrehkopf/sd2snes/issues/186) with the latest SD2SNES firmware, and there are a couple of beta firmwares in the thread for fixing the issue.

I've had mixed success myself. Sometimes the console will boot into 50Hz mode (my menu setting), and correctly swap between 50/60 depending on ROM (with appropriate LED colour change). 95% of the time, it boots into 60Hz with the LED indicator on orange/yellow for "auto" mode (but is in passthrough).

I can confirm the region is automatically being detected with the only other two carts that I have: a PAL Super Mario World, and an NTSC-J Super Game Boy 2. Both flick 50/60 as expected, including LED.

I'm curious if it's a timing issue with my questionable cart. Will see once the FXPAK PRO is received.

jamieshaw, 2024-05-24:
Just to follow up — FXPAK PRO received, swapped the same SD card from the SD2SNES to the FXPAK with ["issue186-try2"](https://github.com/mrehkopf/sd2snes/issues/186#issuecomment-1436132225) firmware and it booted to 50Hz (matching my setting) straight away.  Power cycled a few times, all to green (50Hz).

In comparison, the SD2SNES Revision X would flick to green maybe 1 in 10 attempts.  So for me, yes, a bad "clone" (despite open-source) cart.

I'd definitely recommend using the ["issue186-try2"](https://github.com/mrehkopf/sd2snes/issues/186#issuecomment-1436132225) firmware, if you haven't already.

DerElefant69, 2024-05-24:
Hi, thank you for your informations. My FXPAK PRO is also new. I bought it two months ago. I will try your suggested Firmware when I get home and see If it gets better. Will report back after I tested it.

DerElefant69, 2024-05-24:
Thank you very much! With your suggested firmware it works like a charm now :) So it was a problem with the FXPAK PRO firmware and not the AIO SuperCIC.

closed 2024-05-24 by DerElefant69