borti4938/SNES_MultiRegion_with_DeJitter_QID #20: SFC 1CHIP, cartridge needs Reset Button Press before it boots

retrogamer70, 2024-07-02:
I'm using the old Switchless Modboard (https://github.com/borti4938/SNES_MultiRegion_with_DeJitter_QID/tree/master/old.pcbs/SNES-1Chip) in my Switchless modded 1CHIP Super Famicom.

The console ran perfectly well for the first 5-6 months, I've been playing a lot on it.
But since a couple of weeks, if I want to boot cartridges (or my FXPAK Pro), the console won't output a video signal after I turn it on. The screen just stays black. I already found a workaround: if I press the reset button (short), then the cartridge will boot. Sometimes I have to press the reset button twice, then it boots.

This doesn't annoy me much, since the workaround fixed the "problem" entirely. I was just wondering what kind of problem this could be. I haven't had this behaviour in the first couple of months, and now, all of a sudden this problem exists.

Thanks for and advice or help.

borti4938, 2024-07-25:
Haven't heard about something like that. However, it does not sound that this issue is connected to the region mod. It's more the SNES itself (maybe instable power on power cycle). It's just a wild guess... Have you tried another power brick? Maybe it is also worth to replace the electrolyt capacitors.

retrogamer70, 2024-07-26:
Thanks for your reply and your evaluation. I'll try to find fixes with the console then, rather than with the modboard.